Beckhoff/TF6000_ADS_DOTNET_V5_Samples #4: Cannot connect Server or SymbolicServer examples to specified ports

iplusplus, 2022-01-23:
When starting any of the example server projects, they fail to connect to the specified (hard coded) ports, throwing errors that are catchable (for the Server example) or not catchable (SymbolicServer). I tried setting various port numbers > 0x8000. I also used NETSTAT to check which ports are tied up to make sure I wasn't having clashses.

I changed the ADS.Server example to use the other constructor (takes port name and logger only, NOT a port number as well), which then allows the server to start. There is not an equivalent version of this for the SymbolicServer, both constructors require a port number.

Alternatively, I can specify a port below 0x8000 and it connects.

RalfHeitmann, 2022-01-26:
The last change changing to port 0x8000 was wrong unfortunately.
The TwinCAT Router doesn't accept port numbers >= 0x8000. These are reserved for AdsClients with dynamic port numbers.
User Server Ports must be in between
AmsPort.CUSTOMER_FIRST (25000) <= PORT <= AmsPort.CUSTOMER_LAST (25999)
to not conflict with Beckhoff prereserved servers!
I'll change the examples.

closed 2022-01-26 by RalfHeitmann

odalet, 2022-12-16:
I know this is a closed issue, but I bumped into this recently. Indeed, the samples indicate that the custom port should be in the [25000;25999] range, but using a higher port number used to work seamlessly with the .NET 4 client... 

I have no problem with the new API being breaking, but this breaks assumptions about the protocol behavior. Is there no way the new lib reverted to the old behavior?

commit 69b52074 (2022-01-26): Removing Server Ams ports >= 0x8000 from Samples, fixes #4