Beckhoff/TF6000_ADS_DOTNET_V5_Samples #5: AdsClient crashes on malformed read responses

stegm, 2022-02-17:
We communicate via ADS to an EtherCAT Profibus-Master. For some reason the Master respond sometime with an malformed read response:

[image]

The response set the Length to 240, but actually no data exists in the response.

This works fined in the old V4 version of the ADS client, but the new V5 (5.0.528) reports a ADS timeout and then stops working, until a new client is reconnected.

The log of the ADS client contains the following entry:

```
AmsServer failed with Error: Das angegebene Argument liegt ausserhalb des gültigen Wertebereichs.
Parametername: start  (Port 'AdsClient' (0))!
```

For me it seems that the client tries to access the non-existing data and the crashes somehow.

---

Raw-Data of the response:

[image]

stegm, 2022-02-18:
The problem is that I cannot trigger on the `ArgumentOutOfRangeException` because it is inside the ADS-Dll. On the client side I get the following exception which is obviously not the root cause:

```
   at TwinCAT.Ads.AdsErrorCodeExtensions2.ThrowOnError(AdsErrorCode adsErrorCode)
   at TwinCAT.Ads.AdsClient.Read(UInt32 indexGroup, UInt32 indexOffset, Memory`1 readBuffer)
   at AdsResponseError.Program.Main(String[] args) in c:\temp\AdsResponseError\AdsResponseError\Program.cs:line 25
```

Unfortunately the mentioned log messages does not contain a stack trace either.

But I attached a simple program which uses the `AdsServer` class to produce such a malformed read response message.

[Program.zip](https://github.com/Beckhoff/TF6000_ADS_DOTNET_V5_Samples/files/8096465/Program.zip)

This programm produces the following malformed response which crashes the client.

[image]

RalfHeitmann, 2022-02-21:
Thanks for reporting this. I could detect and fix the problem.
The behaviour of the 5.x will be a little bit different from the 4.x, it will throw an AdsErrorException with AdsErrorCode.ClientResponseInvalid immediatly with receiving the invalid frame.
I assume the 4.x just ignores the erraneous frame and then throws an AdsErrorException with AdsErrorCode.ClientSyncTimout after timeout (what is valid, but suboptimal because the calling thread blocks until timeout).
Please have a look at the upcoming release(s).

closed 2022-02-21 by RalfHeitmann

boj4n, 2022-03-01:
Crashes OnADSNotification first call. _binRead is null

[image]

commit 1813b338 (2022-03-03): Fixing NullReferenceException in BaseSamples 23_TcStateChange, fixes #5