Beckhoff/TF6000_ADS_DOTNET_V5_Samples #9: Add Rpc Method to Server Samples

mark-lazarides, 2022-03-17:
Hi, 

Is it possible to add an Rpc Method to the Ads Symbolic Server samples? I have had a quick look through the docs but can't see how to do that. If you can suggest a way to me, I will try and then add as a PR.

Thanks, Mark

RalfHeitmann, 2022-03-18:
This feature is already on the TODO list and will be implemented in one of the next versions.
The main issue is that we not only have to extend the sample, this feature also needs extensions in the Beckhoff.TwinCAT.Ads package.

mark-lazarides, 2022-03-18:
thanks for the response, @RalfHeitmann. 

Can you guide me on what is necessary? I could put something together to help move it along?

RalfHeitmann, 2022-03-18:
Just be patient ;-)
The plan is to support code like this:

```csharp

PrimitiveType dtInt = new PrimitiveType("INT", typeof(short));

RpcMethod rpc1 = new RpcMethod("Method1",
                new RpcMethodParameterCollection()
                    .AddParameter("i1", dtInt, MethodParamFlags.In)
                    .AddParameter("i2", dtInt, MethodParamFlags.In)
                , dtInt); // Return Type

RpcMethod rpc2 = new RpcMethod("Method2",
                new RpcMethodParameterCollection()
                    .AddParameter("in1", dtInt, MethodParamFlags.In)
                    .AddParameter("out1", dtInt, MethodParamFlags.Out)
                , null); // Return Type

RpcMethodCollection rpcMethods = new RpcMethodCollection()
                .AddMethod(rpc1)
                .AddMethod(rpc2);
                    
RpcStructType dtStructRpc = new RpcStructType("MYRPCSTRUCT", null, members,rpcMethods);
```

closed 2022-04-01 by RalfHeitmann

mark-lazarides, 2022-04-01:
@RalfHeitmann, great, thank you! I will have a play with it this weekend.

commit 564f6e63 (2022-04-01): Add AdsSymbolicServer RpcInvoke Method calls for AdsSymbolicServer, fixes #9,#11,#12