Beckhoff/TF6000_ADS_DOTNET_V5_Samples #17: WhenNotification returns default values

drvic10k, 2022-04-19:
When trying to get the data as an IObservable using this extension, the struct that is returned has all properties with default values

```
public static IObservable<SymbolValueNotification> WhenNotification(
      this IAdsConnection client,
      ISymbol symbol,
      NotificationSettings settings)
```

RalfHeitmann, 2022-04-26:
Unfortunately I couldn't reproduce the issue.
With the following code everything works as expected.

```
TYPE DUT :
STRUCT
	i1 : INT;
	i2 : INT;
END_STRUCT
END_TYPE
```
```
PROGRAM MAIN
VAR
	dut : DUT;
END_VAR

dut.i1 := dut.i1 + 1;
dut.i2 := dut.i1 + 1;
```

``` csharp
using System;
using TwinCAT;
using TwinCAT.Ads;
using TwinCAT.Ads.Reactive;
using TwinCAT.Ads.TypeSystem;
using TwinCAT.TypeSystem;

namespace ConsoleApp1
{
    class Program
    {
        static void Main(string[] args)
        {
            Console.WriteLine("Hello World!");

            using (AdsClient client = new AdsClient())
            {
                client.Connect(851);

                ISymbolLoader loader = SymbolLoaderFactory.Create(client, SymbolLoaderSettings.DefaultDynamic);
                IValueSymbol dutSymbol = (IValueSymbol)loader.Symbols["MAIN.dut"];

                dynamic val1 = dutSymbol.ReadValue();
                var observable = client.WhenNotification(dutSymbol, NotificationSettings.Default);

                observable.Subscribe(n =>
                {
                    dynamic valNot = n.Value;
                    Console.WriteLine(valNot.i1);
                });

                Console.ReadLine();
            }
        }
    }
}
```

Please provide us with more detailed information.

drvic10k, 2022-04-26:
I see that you are using the port 851, which is probably a real server, I am using the AdsSymbolicServer

it seems to be similar to this issue: https://github.com/Beckhoff/TF6000_ADS_DOTNET_V5_Samples/issues/14

RalfHeitmann, 2022-04-27:
Sorry, I thought it could be a bug in the WhenNotification Observer.
I can confirm an issue that doesn't read the correct symbol value in the AdsSymbolicServer when Notifications are fired (Values remain the Default Value). A new version of the packages and sample code is coming soon ...

commit 29183116 (2022-04-27): SymbolicServerSample: Add sample code for Notifications on structures, fixes #17 in combination with Beckhoff.TwinCAT.Ads >= 6.0.94