Beckhoff/TF6000_ADS_DOTNET_V5_Samples #18: SubSymbol access broken for symbols representent by interfaces in ADS 6

maxxie85, 2022-04-28:
When upgrading from ADS 5.0.545 to ADS 6.0.102 SubSymbols don't load anymore in Symbol objects that are accessed via interfaces on the PLC side.

I created a demo project that illustrates this issue. The `Mother` object has 2 `Child` objects that are of type `IChild` (interface). 
On the PC side the Properties of the `Child` objects throws an exception when accessing them. When checking the SubSymbols element of the `Child `the enumaration yields no results. 

However this method was valid and working in ADS 5 but is not working anymore in ADS 6

The PC demo application contains 2 pre-configured Solution configurations named `Working` and `NotWorking` that switches between the 2 ADS versions
[DemoApplication.zip](https://github.com/Beckhoff/TF6000_ADS_DOTNET_V5_Samples/files/8583161/DemoApplication.zip)
.

RalfHeitmann, 2022-04-29:
Hi Maxxie85, 
thanks for reporting this and the demo code.
Had a look inside, but unfortunately this issue is not as easy to solve as it seems on first sight.
The problem has to to with the Inheritance of the 'IChild' type. This type was interpreted as Struct-type in ADS V5 (with its included properties as members) by mistake. The Struct behaviour (accessing the Member properties) is what you are using in your demo code.
But the wrong recognition (Creation of a struct type instead of an InterfaceType, DataTypeCategory.Interface) caused other issues in different situations and we fixed that in ADS V6.
Now the IChild type is recognized properly, but this type actually doesn't support the Generic Member access bound to PLC Properties as the struct type does.
To cut a long story short - there is a little bit of implementation and refactoring work necessary needing a little bit of time ... but we are on-topic!

maxxie85, 2022-05-02:
Hi Ralf,

Thanks for the response. It's great that you guys are willing to pickup the issue. 
We rely heavily on interface access in our software on the PLC side.

Looking forward to the implementation.

RalfHeitmann, 2022-05-04:
Should be fixed with Beckhoff.TwinCAT.Ads 6.0.112

closed 2022-05-04 by RalfHeitmann

maxxie85, 2022-05-04:
Yes working as expected. Thanks