Beckhoff/TF6000_ADS_DOTNET_V5_Samples #21: TwinCAT.Ads Exception: 'Ams Connect: Cannot register Port '0'

emulk, 2022-05-11:
Hello, 
i'm using the version Beckhoff.TwinCAT.Ads 6.0.112, on a .net core 5 project. 
If i run the project on Twincat 3 , and i try to connect with a Twincat 3 plc, evry thing seems good.
When i run my project on Twincat 2.11.0 (Build 2103) i have the following error:

**TwinCAT.Ads Exception: 'Ams Connect: Cannot register Port '0' (Error: Client PortNotOpen)'**

How can i run my project on twincat 2 and connect to a twincat 3 plc.

THank you

RalfHeitmann, 2022-05-12:
Please have a look at the Installation Hints on

https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads

The V5.x and V6.x Versions of the Ads package doesn't support TwinCAT 2 installed on the same system (remote access does work!).
If you want to support TwinCAT 2 locally, you have to use Ads V4.x - which should work also under .NET 5 and .NET 6 (limited for the Windows platform).

closed 2022-05-12 by RalfHeitmann

emulk, 2022-05-14:
Hello,

i can't make it work on .net core 5

i recive the following error when i try V4.x dll
System.BadImageFormatException: 'Could not load file or assembly 'System.Windows.Forms, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089'. Reference assemblies should not be loaded for execution.  They can only be loaded in the Reflection-only loader context. (0x80131058)'

And on the nuget page is clearly indicated that the dll is for - .NET Framework 4.0

do you have any sample of the V4.x dll on .net core 5 project ?

Thank you

RalfHeitmann, 2022-05-23:
Within .NET 5 projects you have to indicate that the project is targeting windows only and you want to run a System.Windows.Forms application (TwinCAT.Ads in V4.X is basing on Windows.Forms).

e.g. in your *.csproj Project File (see the TargetFramework and UseWindowsForms tag)

```csproj
<Project Sdk="Microsoft.NET.Sdk">
  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>net5.0-windows</TargetFramework>
    <UseWindowsForms>true</UseWindowsForms>
  </PropertyGroup>
  <ItemGroup>
    <PackageReference Include="Beckhoff.TwinCAT.Ads" Version="4.4.21" />
  </ItemGroup>
</Project>
```