Beckhoff/TF6000_ADS_DOTNET_V5_Samples #22: Symbol.WhenValuesChanged() throws on TC RT (TCBSD) but works in TC CE7

maxxie85, 2022-05-20:
When registering a WhenValueChanged subscription on a Symbol. The method throws a Parameter value(s) is invalid (ADS error code 1803) exception when using the TwinCAT RTx64 based on TCBSD.

However the same code does works on a CX8190 or CX9020 system.

The code
```C#
Symbol.WhenValueChanged().Subscribe(OnNext)
```

StackTrace
```
TwinCAT.Ads.AdsErrorException: Parameter value(s) is invalid. (AdsErrorCode: 1803, 0x70B)
at TwinCAT.Ads.AdsErrorCodeExtensions2.ThrowOnError(AdsErrorCode adsErrorCode)
at TwinCAT.Ads.AdsClient.AddDeviceNotification(String symbolPath, Int32 dataSize, NotificationSettings settings, Object userData)
at TwinCAT.Ads.AdsConnection.AddDeviceNotification(String variableName, Int32 dataSize, NotificationSettings settings, Object userData)
at TwinCAT.Ads.ValueAccess.AdsValueAccessor.RegisterNotification(ISymbol symbol, SymbolNotificationTypes type, NotificationSettings settings)
at TwinCAT.Ads.ValueAccess.AdsValueAccessor.OnRegisterNotification(ISymbol symbol, SymbolNotificationTypes type, INotificationSettings settings)
at TwinCAT.Ads.TypeSystem.Symbol.add_ValueChanged(EventHandler`1 value)
at MPTCore.TwinCAT.Common.Models.ValueSymbolDecorator`1.add_ValueChanged(EventHandler`1 value) in C:\Source\GIT\MPTCore.TwinCAT\src\MPTCore.TwinCAT\Common\Models\ValueSymbolDecorator.cs:line 190
at TwinCAT.Ads.Reactive.ValueSymbolExtensions.<>c__DisplayClass3_0.<WhenValueChanged>b__0(EventHandler`1 h)
at System.Reactive.Linq.ObservableImpl.ClassicEventProducer`2.AddHandler(TDelegate handler) in /_/Rx.NET/Source/src/System.Reactive/Linq/Observable/FromEvent.cs:line 360
at System.Reactive.Linq.ObservableImpl.EventProducer`2.Session.AddHandler(TDelegate onNext) in /_/Rx.NET/Source/src/System.Reactive/Linq/Observable/FromEvent.cs:line 325
```

maxxie85, 2022-05-20:
Some additional information, I noticed there was a major upgrade for TC/BSD. I was running TC/BSD 12. After the upgrade to TC/BSD 13.0.11.1.2 I still get the same issue.

RalfHeitmann, 2022-05-23:
Just registering the Symbol as Device Notification seems to trigger the problem here. Could you narrow down the problem to a single symbol? Demonstration code would be helpful, C# + PLC.

maxxie85, 2022-05-30:
Well this is most interesting. When I tried to narrow it down, I seem to have solved it myself. Because I can't trigger the exception anymore. Only I thing I can think of is I cleaned up the decorator class we use.

RalfHeitmann, 2022-06-10:
An PLC Property doesn't support Notifications. Therefore a WhenValueChanged should throw an AdsErrorException with the following code
```
/// <summary>
/// Invalid parameter value(s).
/// <para>Error code: 1803 (0x70b).</para>
/// <para>Category: Device</para>
/// <para>C++: ADSERR_DEVICE_INVALIDPARM</para>
/// </summary>
DeviceInvalidParam = 0x0000070b,
```
At least with my actual virtual test system (Windows 64Bit + TwinCAT 4024.23) this works as intended.
I will check this another way, unfortunately I don't have a CX9020 by hand at the moment.

maxxie85, 2022-06-10:
> An PLC Property doesn't support Notifications.

Interestingly that it does works on a CX9020 with TC3.1.4024.10, but that could be a happy accident. 
As it isn't a supported feature, that makes this issue mute. However I do feel that it shouldn't crash the PLC.

RalfHeitmann, 2022-06-10:
Probably it simply neither crashed the Plc in 4024.10, but I personally don't believe that the Notification behind was functional.
Nevertheless It shouldn't crash and It should return a proper AdsErrorException (under WinCE, TcBSD or whatever).
We have to check if the issue is fixed also on ARM platforms - at least with our latest 4024.29.

maxxie85, 2022-06-10:
On WinCE 3.1.4024.10 on the CX9020 it is a functional notification. We have several PLC's on that version where we do our testing and development on. I have a CX8190 also on that version and it also works on that one. That is what confused me the first time, because when I worked on TCBDS it suddenly failed with an exception which I didn't got on the ARM PLC's. 

I can't test the ARM version of 4024.29 as I don't see a public build of that version for the CX9020

RalfHeitmann, 2022-06-10:
You are right, on Win64 3.1.4024.10 everything works as expected, the 3.14024.29 still triggers an AdsErrorException Code 1803 on registering the Notification.

closed 2022-07-19 by RalfHeitmann