Beckhoff/TF6000_ADS_DOTNET_V5_Samples #27: Expose struct fields as separate symbols

drvic10k, 2022-06-27:
How can I expose struct fields as separate symbols, so I can retrieve them directly?

From the AdsSymbolicServer example, I would like to do this:

`$session | read-tcValue -path 'Main.struct1.name'`

But that is not possible.

RalfHeitmann, 2022-06-27:
Actually, the AdsSymbolServer example doesn't support that. The sample lacks code that resolves the filed 'name' from the 'MyStruct' value. The 'Main.struct1' symbol is added to the internal value dictionary as a blob instead.
Because the (Sub)Instance is not found the OnReadRawValue overload returns AdsErrorCode.DeviceSymbolNotFound.

In general it is the SymbolicServer implementation that maps the value to the Symbol Read request and that is not part of the SymbolicServer interface and therefore custom specific.
We don't want to determine an approach. However the sample implementation here in GitHub is actually too simple and we should extend it with a working sample-approach for structs.
This could by a custom value resolver or a different method to expose struct fields in the value dictionary.

drvic10k, 2022-06-27:
could you please point me to the place where I should implement my custom resolver? It seems to fail too soon when I try to retrieve the subinstance. The flow doesn't even reach the AdsSymbolicServer code

RalfHeitmann, 2022-06-28:
In the GitHub Sample it fails in the
SymbolicTestServer::OnReadRawValue method.

https://github.com/Beckhoff/TF6000_ADS_DOTNET_V5_Samples/blob/d80ff8c1b864e05c8aa2a569d40b1226dce0cff5/Sources/ServerSamples/AdsSymbolicServerSample/AdsSymbolicServer.cs#L341

drvic10k, 2022-06-28:
interesting, I will need to have a closer look, because on my implementation, it fails before, I must have missed something

drvic10k, 2022-06-28:
on your example I get `Symbol could not be found (AdsErrorCode: 1808, 0x710)`
but on my example, I get `Instance path not found in InstanceCollection!`

drvic10k, 2022-07-11:
after starting from scratch, I no longer get the error as before, instead I got `Symbol could not be found (AdsErrorCode: 1808, 0x710)` and could implement my own symbol path parsing
but I have no idea, what was wrong before, nevertheless, I think this can be closed

closed 2022-07-11 by drvic10k

RalfHeitmann, 2022-07-11:
Actually, I'm extending the sample that the field access will work already out of the box. Nevertheless the sample code is only a very simplified demonstration that gives you an idea how to bind your own internal value access.
From where the value to get, how it is stored and in which format will be specific in nearly every application.