Beckhoff/TF6000_ADS_DOTNET_V5_Samples #28: ADSClient Connect is successful even if target doesn't exist

enzo-ae, 2022-07-05:
When I run the AdsCli sample, the client only connects to the local router and switches to the Connected state even if the target doesn't exist - is there a way to figure out if the connection actually works or do I check for a variable handle creation to fail?

Tested with Versions 6.0.72(74) and 6.0.155

```csharp
static int Main(string[] args)
{
    args = new[] { "1.1.1.1.1.1:1", "STRING(80)", "CONST.VERSION", "-v" };
    using var loggerFactory = LoggerFactory.Create(builder =>
    {
        builder
            .SetMinimumLevel(LogLevel.Trace)
            .AddConsole();
    });

    ...

    using (AdsClient client = new AdsClient(loggerFactory.CreateLogger<AdsClient>()))

    ...
```

Results:

```diff
2022-07-05T16:22:52.9319048Z: AMS router endpoint set to: 127.0.0.1:48898
2022-07-05T16:22:52.9556310Z: ADS client will connect to ADS service: 1.1.1.1.1.1:1
info: TwinCAT.Ads.AdsClient[0]
      AdsClient created: ID:1, Timeout: 5000 ms
dbug: TwinCAT.Ads.AdsClient[0]
      Connect (Before), Client:AdsClient_1, Address: 1.1.1.1.1.1:1
-info: TwinCAT.Ads.AdsClient[0]
-      OnConnected, Client:AdsClient_1, Address: 1.1.1.1.1.1:1, ClientAddress: 192.168.<AMS Router -> Local Computer -> AMS Net Id>.1.1:<a port in the 30k-s>
-2022-07-05T16:22:53.0325456Z: ADS client connected to ADS service: 1.1.1.1.1.1:1
-info: TwinCAT.Ads.AdsClient[0]
-      AdsClient_1 Connection state changed, NewState: Connected, OldState: Disconnected
2022-07-05T16:22:53.0334660Z: Expected size of type string(80) is 80 bytes
2022-07-05T16:22:53.0337091Z: Will read symbol CONST.VERSION of type string(80) with size 80
2022-07-05T16:22:53.0338165Z: Expected size of type string(80) is 80 bytes
dbug: TwinCAT.Ads.AdsClient[0]
      ReadWriteRequestSync: Addr:1.1.1.1.1.1:1, InvokeId:1, IG:61443:, IO:0, readLen:4, writeLen:e, writeData:43-4F-4E-53-54-2E-56-45-52-53-49-4F-4E-00)
trce: TwinCAT.Ads.AdsClient[0]
      RequestSync: Addr:1.1.1.1.1.1:1, InvokeId:1, serviceId:ReadWrite, Len:e, header:ReadWriteReqHeader(IG:f003,IO:0,ReadLen:4,WriteLen:e, data:43-4F-4E-53-54-2E-56-45-52-53-49-4F-4E-00
dbug: TwinCAT.Ads.AdsClient[0]
      RequestSync: Addr:1.1.1.1.1.1:1, InvokeId:1, serviceId:ReadWrite, Len:e, header:ReadWriteReqHeader(IG:f003,IO:0,ReadLen:4,WriteLen:e, data:43-4F-4E-53-54-2E-56-45-52-53-49-4F-4E-00 succeeded!
dbug: TwinCAT.Ads.AdsClient[0]
      onConfirmationReadWriteAsync: Addr:1.1.1.1.1.1:1, InvokeId:1, Result:TargetMachineNotFound, Length:0, Data:
dbug: TwinCAT.Ads.AdsClient[0]
      RequestAndReceiveSync: InvokeID:'1' succeeded!
2022-07-05T16:22:53.1268910Z: Target machine could not be found. (AdsErrorCode: 7, 0x7)
2022-07-05T16:22:53.1270870Z: Expected size of type string(80) is 80 bytes
2022-07-05T16:22:53.1276700Z: Could not read CONST.VERSION of size 80

info: TwinCAT.Ads.AdsClient[0]
      OnDisconnect: Client:AdsClient_1, Address: 1.1.1.1.1.1:1
info: TwinCAT.Ads.AdsClient[0]
      AdsClient_1 Connection state changed, NewState: Disconnected, OldState: Connected
dbug: TwinCAT.Ads.AdsClient[0]
      AmsServer disconnected. Port 'AdsClient' (0) received 62 bytes!
```

RalfHeitmann, 2022-07-05:
The point is, that the AdsClient is saving communication roundtrips (historically) as-much-as-possible. That means the target is not contacted during Connect. See it as a Connect preparation on the client side only. You have to fire active communication to find out if the target is available (e.g with ReadState or CreateHandler or whatever).
There are thoughts internally to change this behaviour, but that would mean one extra ADS rountrip during Connect. As it is implemented actually, it is up to the user - with the disadvantage of the possibly misleading ConnectionState. As long no Communication error occurred, the ConnectionState stays 'Connected'

RalfHeitmann, 2022-07-19:
Closing this issue because (at least actually) it is 'As designed'.

closed 2022-07-19 by RalfHeitmann