Beckhoff/TF6000_ADS_DOTNET_V5_Samples #35: Incorrect DateTimeOffset value at ValueChangedEventArgs

Barteling, 2022-10-19:
When i receive a value changed notification i want to store the event time.
But the timestamp given with the notification seems to be incorrect. 
It looks likes the time is in UTC 0. But the offset matches my current time zone. 

[image]

RalfHeitmann, 2022-10-21:
Where does the value changed Notification come from. Is it a different system (with different time settings/TimeZone)? Sourcing from a local system, I don't see any difference.

Barteling, 2022-10-21:
Different system, i'll check the time on the system.

Does the date time variable come from the target system (plc)?

RalfHeitmann, 2022-10-21:
Yes, that is the TimeStamp of the Notification occurence on the PLC (target) System.
It bases on the TwinCAT time (that is synchronized with the system time on TwinCAT SystemService Start).

The TwinCAT time could have a drift to the System Time too because it runs independently from the System time between startups.

Barteling, 2022-10-26:
I believe i messed up the date and time myself .
So i close this one

closed 2022-10-26 by Barteling