Beckhoff/TF6000_ADS_DOTNET_V5_Samples #41: TwinCAT.Ads.TcpRouter leaks connections

beppemarazzi, 2022-12-18:
I'm using `AdsRouterConsoleApp` to try out `AdsSymbolicServerSample` both in this repo.

To reproduce the issue:
1. run AdsRouterConsoleApp
2. run AdsSymbolicServerSample (1st time)
3. all works fine
4. stop AdsSymbolicServerSample observing the router console it reports:
```
[11:20:56 ERR] Cannot send data to Pipe 'TwinCAT.Ams.AmsHeader', Cannot access a disposed object.
Object name: 'System.Net.Sockets.TcpClient'.
[11:20:56 WRN] Sending AmsErrorResponse: Error: ClientPortNotOpen, ****.1.1:32787 <-- ****.1.1:25000, Cmd: Write RES, State:MaskAdsResponse, User:167, Bytes:0
```
6. run AdsSymbolicServerSample (2nd time)
7.  the server does not start: it reports `Symbolic Test Server runnning on Address: '0.0.0.0.0.0:0' ...` and the router console reports
```
[11:21:11 WRN] Port '25000' is already in use. Returning Error Code 'PortAlreadyInUse'
[11:21:11 ERR] Cannot send data to Pipe 'TwinCAT.Ams.AmsHeader', Cannot access a disposed object.
Object name: 'System.Net.Sockets.TcpClient'.
[11:21:11 WRN] Sending AmsErrorResponse: Error: ClientPortNotOpen, ****.1.1:32788 <-- ****.1.1:25000, Cmd: Read RES, State:MaskAdsResponse, User:1, Bytes:0
[11:21:11 ERR] Port close response failed with 'PortNotConnected'
```
8. run AdsSymbolicServerSample (3rd time) 
9. all works fine

Recap: the server starts correctly only on odd tryout... 1st, 3rd, 5th!
IMHO there is some leak into the router: it does not correctly release the connection when the first server goes offline. Then the second tryout fails but as side effect it also repair the state releasing the dead connection. Then the 3rd attempt wil success and so on...

Please consider #40 to give us a chanche to better understand what's going on here and maybe help to fix it....

RalfHeitmann, 2023-02-03:
I have implemented several enhancements for this case in 6.0.235. It should work for the standard sample now.
But be aware, that the Router actually is dependant on unregistering the AdsServer (Port 25000) properly (a ADS PortClose message received on Router side). Otherwise the AmsPort keeps blocked.

beppemarazzi, 2023-02-03:
Ok. I'll try the new release and come back with some feedback next week....
Many thanks!!!!!

RalfHeitmann, 2023-09-12:
Closing abandoned issue.

closed 2023-09-12 by RalfHeitmann