Beckhoff/TF6000_ADS_DOTNET_V5_Samples #42: Are the Ads Client and Server implementation in the .NET 6 nugets compatible with TwinCAT 3.1.4024.32 router?

odalet, 2022-12-19:
I'm in the process of porting some ADS client and server code from the legacy v4 to the new v6 nuget packages (exact version is `6.0.216`) and... I have issues with ports:

My initial setup (before porting) is:

* A custom ADS Server is exposed on port 44496
* Some client code connects to this server and writes bytes on an index group/index offset
* My local router is the one provided with Twincat 3.1.4024.32 (on Windows 10 x64):

[image]

Everything works (and did for years).

Now let's say I've ported my client and server code to use v6 nugets.

When running my server I'm greeted with this error:

```
Error starting ADS Server (AdsAms Error: None): AmsConnect: Cannot register Port '44496' (Error: ClientPortNotOpen)
TwinCAT.Ads.Server.AdsServerException: AmsConnect: Cannot register Port '44496' (Error: ClientPortNotOpen)
```

If I change my server port to **25000** (as indicated in Ads Server example code), my server starts, my client connects and everything works again (still using the Twincat 3.1.4024.32 router).

Then I conducted another experiment: I stopped the Twincat 3.1.4024.32 router and instead ran a custom router based on `Beckhoff.TwinCAT.Ads.TcpRouter` nuget (my code is a copy/paste of the sample found in this reposirory)

With this setup (that is: everything built atop the v6 nugets) both my server and client communicate and what's more I can use either the **25000** or the original **44496** port for my server! 

However if I try to use the same client code but compiled against the v4 nuget, things stop working (my write request reaches the server, but it seems the router fails to send the response back to the client...)

Any idea why I'm witnessing all these discrepancies? The ADS/AMS protocol should not have changed or did it? I really wish:

* I could keep using this 44496 port
* Everything would work be it with code compiled against the old v4 or the new v6 packages and with or without the Twincat 3.1 router.

I hope someone can help me on this. Regards.

soberschmidt, 2022-12-19:
Unfortunately you have to change the number. Ports >= 32768 (0x8000) are reserved by the system. Custom server ports can be freely choose within the port range of 26000-26999.

soberschmidt, 2023-01-04:
Port range 25000 - 25999 are reserved for 3th party ADS servers and can be registered at Beckhoff.
Port range 26000 - 26999 can be used in private networks

Ports > 32768 are reserved for ADS clients and are handle differently. Even if you had no issue with v4 in the past. This is not a supported scenario and can lead to other side effects. Please use a port in the range of 26000-26999 or register an official port.

odalet, 2023-01-04:
Hi and thanks for the detailed explanation. Can you point me to some official documentation explicitly explaining those port ranges? I may have missed something but I found nothing of the sort [on the official Beckhoff website](https://infosys.beckhoff.com/index.php).

RalfHeitmann, 2023-02-01:
Unfortunately this isn't documented yet (I actually added the information, it will be online in future).

- In the samples of this Repo
- AmsPort enum documentation 
  [link](https://infosys.beckhoff.com/content/1033/tc3_ads.net/9408352011.html?id=1801810347107555608)

closed 2023-02-01 by RalfHeitmann