Beckhoff/TF6000_ADS_DOTNET_V5_Samples #44: Wrong dependencies in version 6.0.216

drvic10k, 2023-01-25:
In version 6.0.216 you are referencing old version of Abstractions (6.0.190)

[image]

RalfHeitmann, 2023-02-01:
I just checked the dependencies of the ADS Packages. I don't see any wrong dependency there. Which package shows the problem?

[image]
[image]

Eventually you have a problem with your local nuget package cache. You could try to clean it with nuget CLI commands.

drvic10k, 2023-02-01:
it's not visible on the package, it is a runtime exception

RalfHeitmann, 2023-02-01:
Sure, but I expect that within your project a wrong version of the dll is located on the disk, because it is not properly updated by nuget, or because your application build is not renewing the dll from the correct package properly (these are common problems in VisualStudio that bite from time to time).
The runtime simply loads the Assembly it finds in its search path because different Nuget package versions have all the same AssemblyVersion.
You could check the location where the wrong Dll is loaded from in the Vs-Debugger (Modules View) and check that version.

So please ensure:

- remove all TwinCAT.Ads*.dll files from bin and obj folders (or delete these folders) within your VS-Solution
- If that doesn't help delete Nuget package caches and install the Nuget Dependencies again.

Because the package dependencies seem to be correct, it must be an issue in updating/copying the package/dlls on your system. Or picking the wrong dll from an unexpected location during runtime.

drvic10k, 2023-02-07:
the problem was with an old version of one of our own nuget packages having a dependency on 6.0.190

thanks for pointing me to the solution

closed 2023-02-07 by drvic10k