Beckhoff/TF6000_ADS_DOTNET_V5_Samples #45: Wrong complex struct marshalling

drvic10k, 2023-01-25:
I have the following structure:

```    [StructLayout(LayoutKind.Sequential, CharSet = CharSet.Ansi, Pack = 1)]
    public struct SetupAds
    {
        public SetupAds()
        {
        }

        public string MqttIpAddress { get; init; } = "";

        public short LogLevel { get; init; } = 0;

        public string ApiVersion { get; init; } = "";
    }
```

[image]

When I retrieve individual fileds, I get correct values back:

[image]

But retrieving the whole structure is not correct and the string is different every time:

[image]

RalfHeitmann, 2023-02-01:
Sorry, I'm a little bit of context here. You are implementing you own AdsSymbolicServer here? Or what is the situation?

On first sight, hte SetupAds structure is missing the Memory Layout here (and therefore would use Default Marshalling, which doesn't work with dynamic length strings!).

Have you tried to add the StructLayout attribute on your class/struct like this?

```csharp
[StructLayout(LayoutKind.Sequential, CharSet = CharSet.Ansi, Pack = 1)]
struct SetupAds
{
        [MarshalAs(UnmanagedType.ByValTStr, SizeConst = 81)]
        public string MqttIpAddress { get; init; } = "";
        [MarshalAs(UnmanagedType.I2)]
        public short LogLevel { get; init; } = 0;
        [MarshalAs(UnmanagedType.ByValTStr, SizeConst = 81)]
        public string ApiVersion { get; init; } = "";
}
```

You can check the correct marshalling size of your SetupAds struct with

```csharp
int sz = System.Runtime.InteropServices.Marshal.SizeOf(typeOf(SetupAds));
```
which should be 164.

drvic10k, 2023-02-01:
> SetupAds structure is missing the Memory Layout here (and therefore would use Default Marshalling, which doesn't work with dynamic length strings!).

I was not aware of this, I will give it a try
thanks

drvic10k, 2023-02-07:
it works with the additional attributes, thanks

closed 2023-02-07 by drvic10k