Beckhoff/TF6000_ADS_DOTNET_V5_Samples #51: Issues with v 6.0.273

beppemarazzi, 2023-09-21:
In my project I'm referencing Beckhoff.TwinCAT.Ads v. 6.0.273 and there are some strange issues (like hangs during SymbolicServer shutdown).
I've just realized you have unlisted (but not deprecated) exactly that version (see https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads/6.0.273) and I'm going to update quickly my reference to the latest (6.1.74), hoping that it solves my issues...

My question is: is it possible to know why you have delisted v. 6.0.273? Is it possible that my issues are caused by bugs in v.6.0.273 already fixed in latest versions?

RalfHeitmann, 2023-09-21:
Your thinking is correct. There were some issues in shutting down AdsServers in several specific scenarios which should be fixed now. I am actually not fully aware about your scenario, but there is a good chance that it works better now.

About our 'unlisting' strategy:
We are supporting actually 3 different lines of the packages
- Version 4.4.X Series
- Version 6.0.X Series
- Version 6.1.X Series
Because all three lines are mixed up in the Nuget Versions view (lower versions can hardly be found), we unlist older Versions that are replaced by newer versions. This doesn't necessarily mean that the older versions are dysfunctional you can still use it in your existing projects (as long they are not causing issues, so we don't deprecate them). But it makes sense to remove them from the list for new projects or for upgrading.

beppemarazzi, 2023-09-21:
Many thanks

closed 2023-09-21 by beppemarazzi