Beckhoff/TF6000_ADS_DOTNET_V5_Samples #54: Sending data to server after reconnect does not (always) work

jonev-akva, 2024-02-08:
Hi,
This might be related to issue #53, but creating this to showcase that this might be a side-effect of #53. 
As mentioned in #53, there is some inconsistent behavior which means that the error is not reproducible 100% of the time.

[This unit test](https://github.com/jonev-akva/ads-server-reconnect/blob/30823b5ef77c0479f3717c10015ada57fed21544/UnitTest1.cs#L29) showcases the issue.
As mention in the code, this is the following scenario where the issue with reconnection does not work;

1. Client is sending a message which should be directed to the server without errors.
2. The server disconnects
3. Sending an message while the server is disconnected. This has a inconsistent result. Sometimes it returns "NoError" and either Succeeded or Failed. Sometimes it returns Failed and TargetPortNotFound.
4. The server connects again
5. Sending a message to the server.
  - When 3. returns TargetPortNotFound, this also returns TargetPortNotFound - which means that the communication does not work after the reconnect.
 - When 3. returns "NoError", this returns Ok - which means that the communication works after reconnect

In our production environment we will have some network and power loss which means that we need re-connection to work seamlessly.

RalfHeitmann, 2024-02-12:
see answers in #53

jonev-akva, 2024-02-14:
Hi, @RalfHeitmann , thank you for fixing this, the new version of the libraries did resolve this issue in the unit tests. Did you find any correlation between this and #53 ?

RalfHeitmann, 2024-02-14:
The issues are dependent somehow and are - as you stated - fixed with the new version and the given background hints in #53
I close the issues therefore ...

closed 2024-02-14 by RalfHeitmann