Beckhoff/TF6000_ADS_DOTNET_V5_Samples #58: `Cannot access a disposed object` from `AdsRouterConsole`

crishoj, 2024-10-11:
I am using `AdsRouterConsole` in my development environment. 

It works well, but often emits these warnings:

```
fail: TwinCAT.Ads.AdsRouterService.RouterService[0]
      Cannot send data to target '10.3.0.9.1.1:32785 (Endpoint: )' (Command: 10.3.0.20.1.1:853 --> 10.3.0.9.1.1:32785, Cmd:Notification REQ, State:MaskAdsRequest, Err:0, User:0, Bytes:644, Error: Cannot access a disposed object.
      Object name: 'System.Net.Sockets.TcpClient'.)
warn: TwinCAT.Ads.AdsRouterService.RouterService[0]
      Sending AmsErrorResponse: Error: ClientPortNotOpen, 10.3.0.20.1.1:853 <-- 10.3.0.9.1.1:32785, Cmd: Notification RES, State:MaskAdsResponse, User:0, Bytes:0
```

I wonder if this indicates something is wrong in my setup?

RalfHeitmann, 2024-10-14:
Looks like the AdsServer (Port 853) sends an notification and doesn't know about an already Disposed/Disconnected Client (Port 32785) yet. The ClientPortNotOpen response tells the Server that the Client is not available anymore and tells him to stop sending Notifications. This behavior is intended.
To get rid of the warning (and the communication in the background), all Notifications (or Notification Handles) should be unregistered before closing the connection from the client side. Dependent on the API methods you use for registering Notifications, this is done by AdsClients Disconnect/Close/Dispose methods.

crishoj, 2024-10-15:
Thanks for clarifying.

I can add that the client is using the NPM package https://github.com/jisotalo/ads-client with [connection setup 2](https://github.com/jisotalo/ads-client?tab=readme-ov-file#setup-2---connecting-from-unixwindowsetc-system-to-the-plc).