Beckhoff/TF6000_ADS_DOTNET_V5_Samples #59: Random time-outs with PLC

maxxie85, 2024-10-15:
Hi, I have a long running issue where the AdsSession is randomly is experiencing an time-out with the PLC.
For customers the time-out can mean product loss as the PLC will enter a safe state if the communication times-out.

I'm still not sure on the cause

In the attached log file the issue occurs 2 times starting at timestamp:
- 09:19:12.040
- 09:21:41.370

[log20241015.log](https://github.com/user-attachments/files/17375579/log20241015.log)

Are they any common causes for this issue?

RalfHeitmann, 2024-10-18:
A Timeout is exposed on the Client side. What means that the remote device didn't react in the expected time (timeout). This can happen if the target device has a problem (e.g crashed or too high system load), or if the network is unstable.
The communication timeout itself is neither triggering a different state on the PLC nor activates a watchdog. In the most cases the timeout is the symptom not the source.

maxxie85, 2024-10-18:
Hi Ralph,

Thank you for the valuable feedback. 
Is there a way to monitor the cycle exceeds and CPU load via ADS?

closed 2024-10-18 by maxxie85

RalfHeitmann, 2024-11-05:
You could use the Statistics Property of the AdsSession object to get an idea about the number of Errors.
For further analysis I see only the possibilty for Network dumps. Ads Traffic can be dumped with out AdsMonitor Application or with Wireshare - but then there is the Problem to catch the moment where the Problem occurs.
By the way, have you tried to expand the AdsClient.Timeout time? Does this change the situation?

maxxie85, 2024-11-08:
Hi Ralf,

Yes I know about the statistic property.
I think I found the cause of my issues. And they were indeed performance related issues on the PLC. 
Apparently some functions I build hit very heavy on the CX9020, which are most likely related to a memory bottleneck on the system or a ARM limitation. As compared to a CX5110, the same function executes between 4 to 9 times faster on the CX5110 compared to the CX9020. While they both have the same TC performance level 30. 
But the CX5110 has a x86 CPU and utilizes the TwinCAT RT x86 compared to TwinCAT CE (ARMV7) of the CX9020.

But I face a lineair perfomance hit when accessing a array of pointers on the CX9020. The more entries it iterates over how bigger the hit. And it builds quite fast where having 20 items in the array active it could take up to 0.5 milliseconds or 5% on the 10ms PLC cycle task. And if you have multiple instances the load builds up quite fast.

RalfHeitmann, 2024-11-08:
That explains why the AdsClient runs into timeout. 
Unfortunately am not very deep in the realtime behaviour on different platforms and I cannot answer why the ARM is slower.

One option could be to contact our support - we have experts there for this topic.

Another idea because you asked for cycle exceeds:
Probably the following product

https://www.beckhoff.com/en-en/products/automation/twincat/texxxx-twincat-3-engineering/te1010.html
https://infosys.beckhoff.com/content/1033/te1010_tc3_realtime_monitor/index.html?id=4418206447557636659

could help you identifying/optimizing the problem. I am not aware about the costs, neither have I personally used it.

maxxie85, 2024-11-08:
Thanks for the tips. 
I also thought about the new PLC profiler (TE1210) could help me out. But that's only available on 4026 and I'm still on 4024 for the time being.

Another alteration I made, is I followed the best practice for the CX7000 series. Where I slowed down the IO idle task from 1ms to 10ms and changed the priority order that IO Idle task has the lowest priority.

Only I haven't found anything what the impact is of lowering the IO Idle task time other than it frees up the CPU a bit.