Beckhoff/TF6000_ADS_DOTNET_V5_Samples #62: Timing/order issue with multiple clients connecting to different ADS ports

crishoj, 2024-12-14:
Observed a surprising behavior when using `AdsRouterConsole` as a local proxy to connect multiple clients to different target instances on the same IP/AmsNetID.

Example minimal reproduction:

```c#
using TwinCAT.Ads;

var ports = new List<int>
{
    851,
    852,
    853,
    854
};

var clients = new List<AdsClient>();
foreach (var port in ports)
{
    var client = new AdsClient();
    clients.Add(client);
    client.Connect("100.70.46.31.1.1", port);
}

var tasks = clients.Select(client => client.ReadStateAsync(default));

await Task.WhenAll(tasks);

var i = 0;
foreach (var task in tasks)
{
    i++;
    Console.WriteLine($"Task {i} - Success: {task.Result.Succeeded} - Result: {task.Result.State.AdsState.ToString()} - Error: {task.Result.ErrorCode}");
}

foreach (var client in clients) client.Disconnect();
```

Running this, 3 of the 4 tasks fail:

```
Task 1 - Success: True - Result: Run - Error: Succeeded
Task 2 - Success: False - Result: Invalid - Error: ClientSyncTimeOut
Task 3 - Success: False - Result: Invalid - Error: ClientSyncTimeOut
Task 4 - Success: False - Result: Invalid - Error: ClientSyncTimeOut
```

`AdsRouterConsole` outputs:

```
warn: TwinCAT.Ads.AdsRouterService.RouterService[0]
      Connection reset by peer
```

---

When limited to a single client, it works:

```
Task 1 - Success: True - Result: Run - Error: Succeeded
```

The strange thing is: After the first run with a single client, subsequent attempts to connect 4 clients works:

```
Task 1 - Success: True - Result: Run - Error: Succeeded
Task 2 - Success: True - Result: Run - Error: Succeeded
Task 3 - Success: True - Result: Run - Error: Succeeded
Task 4 - Success: True - Result: Run - Error: Succeeded
```

It seems surprising that "warming up" `AdsRouterConsole` with an initial singular connection has an effect.

RalfHeitmann, 2024-12-16:
Tried this out, and I can commit the issue also for the latest versions of the AdsRouterConsole. The issue solved in 6.1.304 is an AdsClient issue that is independent of this problem here. Thanks for reporting.
As you already found out, the issue only appears, when more than one parallel connection is established to a target system initially    the first time. Subsequent connection requests will run without issues ...

RalfHeitmann, 2024-12-16:
I released a fix in Version 6.1.312. Hope that helps.

crishoj, 2024-12-17:
Thanks Ralf!

May I ask you to push the updated source code to this repo, and/or check that the package `Beckhoff.TwinCAT.Ads.AdsRouterConsole` is published on NuGet?

I get `Unable to find package` using `nuget install Beckhoff.TwinCAT.Ads.AdsRouterConsole` (followed instructions on https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads under "Systems without TwinCAT Installation").

jisotalo, 2024-12-17:
@crishoj install the latest `Beckhoff.TwinCAT.Ads.TcpRouter` package to your router project with

`dotnet add package Beckhoff.TwinCAT.Ads.TcpRouter --version 6.1.312`

After that it seems to work as should, great! Same thing with Node.js version.

```c#
Task 1 - Success: True - Result: Run - Error: NoError
Task 2 - Success: True - Result: Run - Error: NoError
Task 3 - Success: True - Result: Run - Error: NoError
Task 4 - Success: True - Result: Run - Error: NoError
```

RalfHeitmann, 2024-12-17:
As @jisotalo stated. The implementation of the Router is in the Beckhkoff.TwinCAT.Ads.TcpRouter package. The AdsRouterConsole is only the host sample application demonstrating the use that package. As starting point use the samples from here:

https://github.com/Beckhoff/TF6000_ADS_DOTNET_V5_Samples/tree/main/Sources/RouterSamples

The AdsRouterConsole package on Nuget is obsolete. It is unlisted already and it shouldn't be found by standard nuget search anymore ...

jisotalo, 2024-12-17:
@RalfHeitmann it seems that the `AdsRouterConsole ` is mentioned multiple times at https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads

There is even an install command for it. That might confuse alot.

crishoj, 2024-12-17:
> The implementation of the Router is in the Beckhkoff.TwinCAT.Ads.TcpRouter package. The AdsRouterConsole is only the host sample application demonstrating the use that package. 

Right, I had forgotten about this distinction. 

After recompiling with the latest `Beckhoff.TwinCAT.Ads.TcpRouter`, the problem is gone. Thanks again for the rapid fix, Ralf!

closed 2024-12-17 by crishoj