Beckhoff/TF6000_ADS_DOTNET_V5_Samples #65: Server disconnects

vvoldsund, 2025-02-26:
Our application has an ADS server that is listening for incomming messages from a TwinCAT client running in a TwinCAT C++ runtime. This application is running on the same machine as the runtime so it uses the installed ADS router for communication (4024.32).

Using version 6.1.298 for the TwinCAT ADS packages, the application will handle that the runtime restarts after the application has started and will continue to receive packages from the runtime. But after upgrading to 6.2.244 this no longer works.

6.1.298:
- Starting application (runtime is running)
- Receiveing packages from runtime
- Restarting runtime
- Receiving packages from runtime

6.2.244:
- Starting application (runtime is running)
- Receiving packages from runtime
- Restarting runtime
- NOT receiving packages from runtime (and the following is printed)

`2025-02-26 08:17:12 +01:00, TwinCAT.Ads.Server.AmsServerImplBase    , "Finishing AmsCommands receiving scheduler"
2025-02-26 08:17:12 +01:00, TwinCAT.Ads.Server.AmsServerImplBase    , "AmsServer disconnected. Port 'RealtimeDataAdsServer' (27004) received 9937241 bytes!"`

RalfHeitmann, 2025-03-03:
Hi vvoldsund,
you could try out the new version 6.2.283. This should solve the problem.

https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads/6.2.283

vvoldsund, 2025-03-05:
@RalfHeitmann thanks for the patch. Unfortunately the patch introduces other errors.

Our application uses the local router if the application is running on a machine with TwinCAT installed, if not it will use the 'AmsTcpIpRouter'. 'StartAsync' of 'AmsTcpIpRouter' now fails, the router state goes: 'starting' -> 'stopping' -> 'started' -> 'stopped'.

The following error is printed:
`[2025-03-05 13:36:00 ERR] Backend - TwinCAT.Ads.TcpRouter.AmsTcpIpRouter
    Only one usage of each socket address (protocol/network address/port) is normally permitted.`

One route is added in appsettings.json and there are multiple network interfaces that it starts listening on. 6.2.244 did not have this issue.

Also 'ConnectServer' on 'AdsServer' fails. This is probably expected as the router does not start, but the exception makes no sence:
`System.Exception: AmsServer 'AdsOverMqtt' is not set!`

RalfHeitmann, 2025-03-06:
This seems to be the same as issue #66.
A fix is on the way.

> System.Exception: AmsServer 'AdsOverMqtt' is not set!

As you remarked this is expected. When standard TcpIp communication fails, there is a new fallback trying to use an AdsOverMqtt fallback (new in Version 6.2). This fails also because no MqttBroker is configured.
I will enhance the error message.

RalfHeitmann, 2025-03-10:
Please try out the version 6.2.296
https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads/6.2.296

Ralf

vvoldsund, 2025-03-17:
@RalfHeitmann I have now tried 303 and it is not working in the same way as before.

In the setup where the application uses the local router and the TwinCAT C++ runtime is running on the same machine we used to use the localhost address (127.0.0.1.1) to communicate between the two, this seems to be broken. If in my TwinCAT C++ runtime instead use the address of the IPC it seems to be working.

RalfHeitmann, 2025-03-17:
Sorry, I am completely lost about your scenario here right now.
What do you mean with TwinCAT C++ runtime and with runtime in common? Please give me an overview about what concrete scenario is broken (Information about running AdsClients, AdsServers and TcpIpRouters/TwinCAT Routers) and furthermore their network parametrizations (used AmNetIds, AmsPorts, TcpIp Addresses/ Ports if changed. Why are you using a localhost Address for communication?

vvoldsund, 2025-03-18:
@RalfHeitmann the application that is running this package is running at the same computer as the TwinCAT C++ XAR and TwinCAT router. In the TwinCAT C++ XAR there is a client that publishes data to 127.0.0.1.1.1 on a specified port number. The application (server) listens on the specified port number. In 6.1.298 the application would receive the data from the client, in 6.1.303 it does not receive the data.

We are using localhost for the communication because the client and server is running on the same machine, it is robust and requires no configuration. It has been working this way for the last seven years or so.

RalfHeitmann, 2025-03-18:
So, I repeat what I understand by now:
- you run TwinCAT XAR (4024.32) on the local system (so there is **no TcpIpRouter component running in that scenario**, therefore no need for a specific Router/TcpIp configuration)
- The addressed AdsServer (RealtimeDataAdsServer on custom AmsPort 27004) is stable / not changed with newer Ads Packages
- Upgrading C# AdsClient form **6.1.298** to **6.2.303** (6.1.303 was never released!) breaks your application (no communication between AdsClient and AdsServer)
- you use AmsNetId 127.0.0.1.1.1 for the communication between AdsClient and AdsServer

The error you first reported should be fixed with Version 6.2.303 (you confirmed that):
`2025-02-26 08:17:12 +01:00, TwinCAT.Ads.Server.AmsServerImplBase    , "Finishing AmsCommands receiving scheduler" 2025-02-26 08:17:12 +01:00, TwinCAT.Ads.Server.AmsServerImplBase    , "AmsServer disconnected. Port 'RealtimeDataAdsServer' (27004) received 9937241 bytes!"
`

The following error also should be fixed with Version 6.2.303:
`[2025-03-05 13:36:00 ERR] Backend - TwinCAT.Ads.TcpRouter.AmsTcpIpRouter Only one usage of each socket address (protocol/network address/port) is normally permitted.
`

Do you have an error message/log output for your actual still existing problem?

Ralf

vvoldsund, 2025-03-18:
Yes that is correct, the first two reported errors are fixed in 6.2.303, but there is the new error that the server does not receive the data that is sent on localhost. No errors or log on the server side to indicate that anything is wrong. Package is just not received.

RalfHeitmann, 2025-03-18:
Is it possible for you to try out Microsoft Powershell (pwsh) with our Module 'TcXaeMgmt'? This uses the same Ads packages in the background (TcXaeMgtm 6.2.117 uses internally Ads components Version 6.2.246)

https://www.powershellgallery.com/packages/TcXaeMgmt/6.2.117

If I try this here, I get communication via 127.0.0.1.1.1 Port 10000 without problems
```
PS> $s = new-tcSession -NetId 127.0.0.1.1.1 -port 10000
PS> $s | get-adsstate

Target               NetId             Port   State      Latency
                                                          (ms)
------               -----             ----   -----      -------
RALFH04              127.0.0.1.1.1     10000  Config     15

```

You could try it with your port 10000/27004. If the returned state is not 'Invalid' then the AdsServer receives Requests and answers successfully.

RalfHeitmann, 2025-03-19:
One piece of the puzzle is that the AdsServer (Port 26000) shouldn't answer with a destination address of
'''
DST:127.0.0.1.1.1:27002
'''
on a Request with Source 172.0.0.1.1.1 but should replace the NetId to the real NetId (for security reasons). Other AdsServer behave different.
How are the Servers 26000 and 27002 implemented? Both with the Beckhoff.TwinCAT.AdsServer C# Api? Or other Api/Sdk?

vvoldsund, 2025-03-19:
Using 6.1.298 the ADS server do respond to a localhost address as seen in the AmsAdsViewer.
[image]

Upgrading to 6.2.303 it no longer respond to the localhost address as seen in the AmsAdsViewer.
[image]

The client (using port 26000 in this case) is a ADS client in the TwinCAT C++ XAR version 4024.32. Using AdsReadReq with AMSNETID_LOOPBACK. The constraint you mention that the server should not respond to localhost is a new one, it has not been like that before.

RalfHeitmann, 2025-03-20:
Thx, I see it much clearer now.  The problem is on the AdsServer side not allowing Requests targeted to 172.0.0.1.1.1 anymore (as the log and the Ams protocol shows).
Package version 6.2.313 
[NugGet](https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads.Server/6.2.313)
allows these Requests again.

vvoldsund, 2025-03-24:
Thanks, seems to be connecting and reconnecting correctly in 6.2.313.

closed 2025-03-24 by vvoldsund