Beckhoff/TF6000_ADS_DOTNET_V5_Samples #66: `AmsTcpIpRouter` binding to wrong loopback address

crishoj, 2025-03-06:
I am having trouble getting the latest `Beckhoff.TwinCAT.Ads.TcpRouter` (6.2.283) to bind the loopback connection to the right address.

Minimal `Worker.cs`, based on https://github.com/Beckhoff/TF6000_ADS_DOTNET_V5_Samples/tree/main/Sources/RouterSamples/AdsRouterConsoleApp:

```cs
using System;
using System.IO;
using System.Text;
using System.Threading;
using System.Threading.Tasks;
using Microsoft.Extensions.Configuration;
using Microsoft.Extensions.Hosting;
using Microsoft.Extensions.Logging;
using TwinCAT.Ads.TcpRouter;

namespace AdsRouterConsoleApp
{

    public class RouterService : BackgroundService
    {
        private readonly ILogger<RouterService> _logger;
        private readonly ILoggerFactory _loggerFactory;
        private readonly IConfiguration _configuration;

        public RouterService(ILoggerFactory loggerFactory, IConfiguration configuration)
        {
            _loggerFactory = loggerFactory;
            _configuration = configuration;
        }

        protected override async Task ExecuteAsync(CancellationToken cancel)
        {
            Console.WriteLine("Press Ctrl + C to shutdown!");
            
            var router = new AmsTcpIpRouter(_configuration, _loggerFactory);
            Task routerTask = router.StartAsync(cancel);
            
            await Task.WhenAll(routerTask);
            Console.WriteLine("Finished");
        }

    }
    
}
```

`appSettings.json`:

```json
{
  "AmsRouter": {
    "Name": "LocalSystem",
    "NetId": "100.117.55.75.1.1"
  },
  "Logging": {
    "LogLevel": {
      "Default": "Debug",
      "Microsoft": "Warning"
    },
    "Console": {
      "IncludeScopes": true
    }
  }
}
```

This gives an `Address already in use` error, seemingly because `AmsTcpIpRouter` tries to listen on the same interface for both external and loopback connections:

```
dbug: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Configured Configuration Providers:
dbug: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Provider: JsonConfigurationProvider
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Allowed EXTERNAL LOOPBACK NETWORK: 0.0.0.0/0
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Local System Name: LocalSystem
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Local AmsNetId:    100.117.55.75.1.1
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      IPAddresses:       192.168.0.128
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      External Port:     bf02
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Loopback IP:       127.0.0.1
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Loopback Port:     bf02
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      LoopbackExternals: 0.0.0.0/0
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Configured routes:
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      ==================
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      AmsTcpIpRouter Status changed to 'Starting'
dbug: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Starting remote listeners for 'External' ...
dbug: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Starting remote listening on 192.168.0.128:48898, DebugHint: External
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Listening to external Endpoint: 192.168.0.128:48898 (DebugHint: External_192.168.0.128:48898).
dbug: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Starting Listener 'External_192.168.0.128:48898'
fail: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Address already in use
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Receiving from Listener 'External_192.168.0.128:48898' () is cancelled! Stopping receiving ...
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Finishing Listening on AmsTcpIpRouter (NetId:100.117.55.75.1.1, LoopbackEndpoint:127.0.0.1:48898, ExternalPort:48898)
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Finishing listening on (external) Endpoint: 192.168.0.128:48898
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      AmsTcpIpRouter Status changed to 'Stopping'
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Loopback listener 'Loopback' (Endpoint: 192.168.0.128:48898) stopped!
```

If I add `"LoopbackPort": 48800` to the config to force another port, the error is gone, but the loopback listener is still bound on the external interface (`192.168.0.128`):

```
dbug: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Starting remote listeners for 'External' ...
dbug: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Starting remote listening on 192.168.0.128:48898, DebugHint: External
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Listening to external Endpoint: 192.168.0.128:48898 (DebugHint: External_192.168.0.128:48898).
dbug: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Starting Listener 'External_192.168.0.128:48898'
dbug: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Loopback Listener 'Loopback' (192.168.0.128:48800) starting ...
dbug: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Starting Listener 'Loopback_192.168.0.128:48800'
```

Perhaps I am doing something completely wrong?

crishoj, 2025-03-06:
I also tried explicitly adding `"Loopback": "127.0.0.1"` to the config, with the same result.

RalfHeitmann, 2025-03-06:
> info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
>       External Port:     bf02
> info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
>       Loopback IP:       127.0.0.1
> info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
>       Loopback Port:     bf02
> info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
>       LoopbackExternals: 0.0.0.0/0
> dbug: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
>       Starting remote listening on 192.168.0.128:48898, DebugHint: External
> info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
>       Listening to external Endpoint: 192.168.0.128:48898 (DebugHint: External_192.168.0.128:48898).
> dbug: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
>       Starting Listener 'External_192.168.0.128:48898'
> fail: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
>       Address already in use

Hi crishof,
the configuration looks ok. Internal (loopback) communications go to 127.0.01:48898 and External communication port is 172.168.0.128:48898.
That should be the default setting.
Something else seems to block the External Port. Is there any concurrent TwinCAT Router running on that system using the same IPAddresses and ports (e.g. a standard TwinCAT installation)?.

You coud Start/Stop TwinCAT with:

net stop TcSysSrv
net start TcSysSrv

If you want to use the TcpIPRouter in parallel to a running TwinCAT Installation you need to change the Ports on the Loopback side and the External side.

crishoj, 2025-03-06:
TwinCAT is not installed on the machine — we're trying to run a standalone ADS router for a client application.

The thing that stands out in my eyes is this line:

```
      Loopback listener 'Loopback' (Endpoint: 192.168.0.128:48898) stopped!
```

That sounds like the loopback listener attempted to use `192.168.0.128` instead of `127.0.0.1`, even though `127.0.0.1` was announced as `Loopback IP` during startup:

```
info: TwinCAT.Ads.TcpRouter.AmsTcpIpRouter[0]
      Loopback IP:       127.0.0.1
```

crishoj, 2025-03-06:
Also works in 6.2.244, so from my perspective it appears the issue was introduced in 6.2.283. 
Hope this helps 🙂

crishoj, 2025-03-09:
Thanks, Ralf. 
Works well in version 6.2.296.

closed 2025-03-09 by crishoj