Beckhoff/TF6000_ADS_DOTNET_V5_Samples #67: ADS Connection from MacOS to PLC fails

patdhlk, 2025-03-10:
I want to connect from a MacOS system to a remote TwinCAT PLC running `3.1.4024.60`

dotnet SDK: `v8.0.203`
MacOS version: `Sequoia Version 15.3.1 (24D70)`
Library versions: `Beckhoff.TwinCAT.Ads v6.2.296`
PLC: `Windows 10 Enterprise LTSC 10.0.17763 Build 17763`

`client.Connect()` fails with the following exception: 

```
System.NotSupportedException: Internal RouterSockets couldn't be created on this system!
   at TwinCAT.Ads.SocketFactory.CreateConnectedSocketAsync(ILogger logger, CancellationToken cancel)
   at TwinCAT.Ads.SocketFactory.CreateAndConnectAsync(ILogger logger, CancellationToken cancel)
   at TwinCAT.Ads.SocketFactory.<>c__DisplayClass0_0.<<CreateAndConnect>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at TwinCAT.Ads.AsyncHelper.RunSync[TResult](Func`1 func)
   at TwinCAT.Ads.SocketFactory.CreateAndConnect(ILogger logger)
   at TwinCAT.Ads.Server.AmsServerNet.RegisterLoopbackClient()
   at TwinCAT.Ads.Server.AmsServerNet.OnAmsConnect(UInt16 port, String portName)
   at TwinCAT.Ads.Server.AmsServerImplBase.AmsConnect(UInt16 port, String portName)
   at TwinCAT.Ads.Server.AmsServer.ConnectServer()
   at TwinCAT.Ads.Server.AdsServer.ConnectServer()
   at TwinCAT.Ads.AdsClient.Connect(AmsAddress address)
```

I'm running this sample as AdsRouter: https://github.com/Beckhoff/TF6000_ADS_DOTNET_V5_Samples/tree/main/Sources/RouterSamples/AdsRouterConsoleApp

I tried it with the `appSettings.json` and hardcoded values. Do I missing something? According to the documentation, it is supported. Please reach out if additional information is required.

RalfHeitmann, 2025-03-11:
Yes there was missing a MacOS handling internally. I fixed that for version 6.2.301 (should now be available). Please be aware that your scenario 'should' work on MacOS because the support of .NET is given, but actually we don't have the ability to test  for MacOS specifically ...

patdhlk, 2025-03-11:
@RalfHeitmann great news! Version `6.2.301` fixed the connection problem, it's working perfectly now. And, if you're looking for MacOS beta testers, I'm your guy! Feel free to reach out.

closed 2025-03-11 by patdhlk