Beckhoff/TF6000_ADS_DOTNET_V5_Samples #68: System.OverflowException: Arithmetic operation resulted in an overflow - leads to an unhandled exception

Stiwa-PohG, 2025-03-18:
Our application has an ADS server that is listening for incoming messages from a TwinCAT runtime (TC3). The application runs not on the same machine as the TwinCAT runtime.

The machine where the application runs:
* WIN10
* TwinCAT System Service v3.1.0.2418
* TwinCAT v3.1.4024.7
* Beckhoff.TwinCAT.Ads 6.2.244

The application runs for a while without any problems and after an undefined time the following error occurs:
- >System.OverflowException: Arithmetic operation resulted in an overflow
   - Method: _private IObservable<AmsCommand?> createReceiverObservable(string debugHint)_ in the class **AmsServerImplBase**
   - debugHint: "AmsConnectResponse_AdsClient_44730"

It is not possible to catch the exception and so the application **terminates**!

[image]

[image]

[image]

RalfHeitmann, 2025-03-18:
The Exception seems to happen in the AdsClients internal receive observable. Having a short look at the relevant source code, I found one suspicious location that could result into an int32 overflow. This could appear when already 2GB data is received by this AdsClient or AdsServer. Could this be the case here?

Stiwa-PohG, 2025-03-18:
Which variable in which class could I use to determine this in the event of an error/operation?

RalfHeitmann, 2025-03-18:
We don't have that data for public access (just internal variables yet). The intention of the question was that you can double check if my assumption would fit in your observations.

You stated:
"after an undefined time the following error occurs:"

2GB would take some time (depending on your applications network access). Are we talking about seconds, hours or days?

vvoldsund, 2025-03-18:
Not the same exeption, but the same result, application crashed.

From event log:
`Application: AKVAgroup.AKVAconnect.Backend.Service.exe
CoreCLR Version: 8.0.1124.51707
.NET Version: 8.0.11
Description: The process was terminated due to an unhandled exception.
Exception Info: System.NullReferenceException: Object reference not set to an instance of an object.
   at TwinCAT.Ads.Server.AmsServerNet.<>c__DisplayClass58_0.<createReceiverObservable>b__0(IObserver`1 ob)
--- End of stack trace from previous location ---
   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow(Exception exception)
   at System.Reactive.ExceptionHelpers.Throw(Exception exception)
   at System.Reactive.Stubs.<>c.<.cctor>b__2_1(Exception ex)
   at System.Reactive.AnonymousSafeObserver`1.OnError(Exception error)
   at System.Reactive.Sink`1.ForwardOnError(Exception error)
   at System.Reactive.ObserveOnObserverNew`1.DrainShortRunning(IScheduler recursiveScheduler)
   at System.Reactive.Concurrency.EventLoopScheduler.Run()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
 `

- TwinCAT System Service v3.1.0.2437
- TwinCAT v3.1.4024.32
- Beckhoff.TwinCAT.Ads 6.1.260

RalfHeitmann, 2025-03-18:
We have published a version with a potential fix (see above) for the System.OverflowException (Version 6.2.307). You could try that out and check if the error is gone ...

Stiwa-PohG, 2025-03-19:
i tried the new version 6.2.307 and the exception still occurs

**Logging:**
_OnReceiveAmsCommands: TcpHeader with 6 bytes received
AmsServerImplBase:createReceiverObservable Exception: System.OverflowException: Arithmetic operation resulted in an overflow.
   at TwinCAT.Ads.Server.AmsServerNet.OnReceiveAmsCommands(IObserver`1 ob, String debugHint)
   at TwinCAT.Ads.Server.AmsServerImplBase.<>c__DisplayClass65_0.<createReceiverObservable>b__0(IObserver`1 ob)!
AmsServerImplBase(ServerAddress:248.117.164.1.80.245:32930,Hint:AmsConnectResponse_AdsClient_32930: Observer Finished
Finishing AmsCommands receiving scheduler
AmsServer disconnected. Port 'AdsClient' (32930) received 2789000 bytes!_

**Exception stacktrace:**
_System.OverflowException: Arithmetic operation resulted in an overflow.
   at TwinCAT.Ads.Server.AmsServerNet.OnReceiveAmsCommands(IObserver`1 ob, String debugHint)
   at TwinCAT.Ads.Server.AmsServerImplBase.<>c__DisplayClass65_0.<createReceiverObservable>b__0(IObserver`1 ob)
--- End of stack trace from previous location ---
   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow(Exception exception)
   at System.Reactive.ExceptionHelpers.Throw(Exception exception)
   at System.Reactive.Stubs.<>c.<.cctor>b__2_1(Exception ex)
   at System.Reactive.AnonymousSafeObserver`1.OnError(Exception error)
   at System.Reactive.Sink`1.ForwardOnError(Exception error)
   at System.Reactive.ObserveOnObserverNew`1.DrainStep(ConcurrentQueue`1 q)
   at System.Reactive.ObserveOnObserverNew`1.DrainShortRunning(IScheduler recursiveScheduler)
   at System.Reactive.ObserveOnObserverNew`1.<>c.<.cctor>b__17_0(IScheduler scheduler, ObserveOnObserverNew`1 self)
   at System.Reactive.Concurrency.ScheduledItem`2.InvokeCore()
   at System.Reactive.Concurrency.ScheduledItem`1.Invoke()
   at System.Reactive.Concurrency.EventLoopScheduler.Run()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)_

RalfHeitmann, 2025-03-20:
Ok, so my first bet was wrong, but with the logging we seem to be near.
My next assumption is, that the incoming TcpFrame is corrupt. I have enhanced the logging producing an error/warning in that case. 

That wouldn't fix the problem but could help to detect the reason ...

Please try to reproduce the error with this version
[Version 6.2.313](https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads/6.2.313)
and send me a log as above.

Stiwa-PohG, 2025-03-27:
Application runs until now and no error occurred. I think your changes solved the problem.
I will continue to observe it anyway

closed 2025-04-14 by RalfHeitmann