Beckhoff/TF6000_ADS_DOTNET_V5_Samples #69: Unhandled "System.NullReferenceException" crashes application

vvoldsund, 2025-03-24:
Unhandled exception crashes application the application.

TwinCAT System Service v3.1.0.2437
TwinCAT v3.1.4024.32
Beckhoff.TwinCAT.Ads 6.1.260

From Windows event log:
Application: ***.Service.exe CoreCLR Version: 8.0.1124.51707 .NET Version: 8.0.11 Description: The process was terminated due to an unhandled exception. Exception Info: System.NullReferenceException: Object reference not set to an instance of an object. at TwinCAT.Ads.Server.AmsServerNet.<>c__DisplayClass58_0.<createReceiverObservable>b__0(IObserver1 ob)
--- End of stack trace from previous location ---
at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow(Exception exception)
at System.Reactive.ExceptionHelpers.Throw(Exception exception)
at System.Reactive.Stubs.<>c.<.cctor>b__2_1(Exception ex)
at System.Reactive.AnonymousSafeObserver1.OnError(Exception error) at System.Reactive.Sink1.ForwardOnError(Exception error)
at System.Reactive.ObserveOnObserverNew1.DrainShortRunning(IScheduler recursiveScheduler) at System.Reactive.Concurrency.EventLoopScheduler.Run() at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state) --- End of stack trace from previous location --- at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)

RalfHeitmann, 2025-03-24:
The Beckhoff.TwinCAT.Ads Version 6.1.260 is from August 2024. If you have issues please use the actual 6.2.313. In case you have compatibility issues the 6.1.332 is an option (there are 1 or 2 minor breaking changes from 6.1 to 6.2). Is the NullReferenceException still appearing in newer Versions?

vvoldsund, 2025-03-25:
Unfortunately, we have only seen this crash in production, and it happens very sporadically. It may take some time before the next release is deployed to production, so it’s difficult to confirm if the issue persists in version 6.2.313 at this point.

From one of the systems, I managed to capture an error with a bit more context, which references _createReceiverObservable_:

```
Application: ***.Service.exe
CoreCLR Version: 8.0.1124.51707
.NET Version: 8.0.11
Description: The process was terminated due to an unhandled exception.
Exception Info: System.NullReferenceException: Object reference not set to an instance of an object.
   at TwinCAT.Ads.Server.AmsServerNet.<>c__DisplayClass58_0.<createReceiverObservable>b__0(IObserver`1 ob)
--- End of stack trace from previous location ---
   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow(Exception exception)
   at System.Reactive.ExceptionHelpers.Throw(Exception exception)
   at System.Reactive.Stubs.<>c.<.cctor>b__2_1(Exception ex)
   at System.Reactive.AnonymousSafeObserver`1.OnError(Exception error)
   at System.Reactive.Sink`1.ForwardOnError(Exception error)
   at System.Reactive.ObserveOnObserverNew`1.DrainShortRunning(IScheduler recursiveScheduler)
   at System.Reactive.Concurrency.EventLoopScheduler.Run()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
```

Would you expect any changes in 6.2.313 that specifically address this scenario, or would it require deeper code examination to identify the root cause?

RalfHeitmann, 2025-03-25:
Analysing the source code (Version 6.1.260) I found one issue in the createReceiverObservable that is suspicious for a NullReferenceException. But that's a very rare edge case I actually don't know it can happen at all. So it's very vague that this is your issue. But anyway, this exists still in the actual 6.1 and 6.2 versions and this is fixed for future versions (6.1 and 6.2).
In our tests we didn't experience a NullReferenceException after Version 6.1.260 and therefore there is no explicit fix for it 6.1 or 6.2.
Tell me if you need a new Version 6.1 for testing ...

vvoldsund, 2025-03-28:
Thanks for looking into this. Our next release will be using 6.2.313 or newer, so there's no need for a 6.1 test version. Unfortunately, it will be a while before that version is released, but I will monitor the situation and provide feedback if this occurs again.

closed 2025-05-09 by RalfHeitmann