Beckhoff/TF6000_ADS_DOTNET_V5_Samples #71: TcpHeader indicates more than '4115661220' bytes of data! Fatal Error

Stiwa-PohG, 2025-04-30:
What does this error mean?

[ERROR] [TwinCAT.Ads.Server.AmsServerImplBase] OnReceiveAmsCommands: TcpHeader invalid. Dump: TcpHeader Flags: 30200, DataSize: 4115661220
[FATAL] [TwinCAT.Ads.Server.AmsServerImplBase] OnReceiveAmsCommands: TcpHeader indicates more than '4115661220' bytes of data! Fatal Error.

I'm using the the following versions:

WIN 11
Beckhoff.TwinCAT.Ads 6.2.313
thx

RalfHeitmann, 2025-05-05:
These are the extended logging messages that are added to analyse your problem further (see issue #68).
Unfortunately the TcpFrame that should be received here seems to be defect. The included TcpHeader is already nonsense (the delivered flags are unknown and the data load indicates over 4Gigs). At this point, the TcpConnection cannot be recovered and the AdsClient is broken - and it is not clear which side of the TcpConnection causes the trouble (the AdsClient, the AdsServer or the TwinCAT Router in-between).

Some further questions:
- Is there any AdsClient log available for the time just before the problem?
- This seems to be a different system than described in #68. Is TwinCAT v3.1.4024.7 still the realtime system (whole system running locally?). Have you tried a more actual TwinCAT Version?
- About (your own) AdsServer you mentioned. What is the implementation framework (Plc, C++, Beckhoff.TwinCAT.AdsServer package or whatever). 

Because this seems to be a multi discipline problem that needs a deeper inspection it will be helpful to proceed with the help of our regional support.

In the meanwhile we don't have the root of the problem, we will enhance the logging and try to implement more sophisticated fallback/error strategies for this specific error case.

RalfHeitmann, 2025-06-11:
You could try out Version 6.2.373 which has some significant fixes and extended logging that could help with this issue:
https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads.ConfigurationProviders/6.2.373

RalfHeitmann, 2025-07-21:
Haven't got any reports about this the recent weeks. I assume the issue is fixed. Please reopen or create a new issue if it happens again.

closed 2025-07-21 by RalfHeitmann