Beckhoff/TF6000_ADS_DOTNET_V5_Samples #74: Beckhoff.TwinCAT.Ads.ConfigurationProviders key already added exception.

sdijoseph, 2025-06-09:
When using version 6.2.347 of the Beckhoff.TwinCAT.Ads.ConfigurationProviders package, the method AddStaticRoutesXmlConfiguration fails with this exception:

```
System.ArgumentException: An item with the same key has already been added. Key: $AmsRouter:RemoteConnections:{i}:NetId:Tls.Cert
   at System.Collections.Generic.Dictionary`2.TryInsert(TKey key, TValue value, InsertionBehavior behavior)
   at System.Collections.Generic.Dictionary`2.Add(TKey key, TValue value)
   at TwinCAT.Ads.Configuration.StaticRoutesXmlConfigurationProvider.Load()
   at Microsoft.Extensions.Configuration.ConfigurationManager.AddSource(IConfigurationSource source)
   at Microsoft.Extensions.Configuration.ConfigurationManager.ConfigurationSources.Add(IConfigurationSource source)
   at Microsoft.Extensions.Configuration.ConfigurationManager.Microsoft.Extensions.Configuration.IConfigurationBuilder.Add(IConfigurationSource source)
   at TwinCAT.Ads.Configuration.StaticRoutesXmlConfigurationBuilderExtension.AddStaticRoutesXmlConfiguration(IConfigurationBuilder builder, ILoggerFactory logger)
```

In the decompiled code, I see these two lines called right next to each other, I think it's likely they are the cause.

```
          this.Data.Add("$AmsRouter:RemoteConnections:{i}:NetId:Tls.Cert", trustedRoute.Tls.Cert);
          this.Data.Add("$AmsRouter:RemoteConnections:{i}:NetId:Tls.Cert", trustedRoute.Tls.Cert);
```

Please let me know if I'm overlooking something.

RalfHeitmann, 2025-06-11:
The issue should be fixed in Version 6.2.373:
https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads.ConfigurationProviders/6.2.373

sdijoseph, 2025-06-11:
> The issue should be fixed in Version 6.2.373: https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads.ConfigurationProviders/6.2.373

And so it is! Thanks!